andreasdahl1987/DahlDesignDDC #21: Bite button point with paddle feature

Rekiem24, 2023-03-26:
Hi!

I'll try to explain the main issue using the bitebutton() function and dual clutch.

Slave clutch values are min: 18 and max 138

When I hold the slave or master clutch to desired bite point, then press bitebutton() it sets correctly the bite point %.

Once set the bite point, release the clutch, then bitebutton should work as normal push button, however when I press this button it change the bite point % to 0.

Also this happens with bitebutton() and rotary2bite(). I can set the first +/-10 however when pressing bitebutton() to change state for +/-1.....the bite point % goes to 0.

Is it possible that bcs I have my slave clutch set to 18 the code thinks the clutch is slightly pressed? or bitebutton() cannot work as a normal button?

Thanks!

Awesome work!

andreasdahl1987, 2023-03-26:
Hi! Yes I suspect one of the clutches arent stable at 0%. The feature is handeled [here](https://github.com/andreasdahl1987/DahlDesignDDC/blob/02b93da7ac60951cc820121472975ef8d8ab8bf7/DahlDesignDDC/81_DualClutch.ino#L112). The button wont set the bite point directly if both are at 0%. So try making an adjustment to the "released" value and see if that helps. Changing clutch modes also requires a stable 0% and being able to reach 100%, so you might have issues there as well? 

If you get it working but are unhappy with the result(jittery or deadzone too big) you can edit the function in the link I sent. The value 0 can be set to for instance 100, so that you cant set bite point unless a paddle is at least 10% pressed

Rekiem24, 2023-03-26:
Hi!

Thanks! fixed the issue adjusting the "released" parameters. 

This was the first option however it seems I wasn't adjusting enough or maybe the deadzone is too big

Thanks

closed 2023-03-26 by Rekiem24