andreasdahl1987/DahlDesignDDC #22: LED or Power Control using PWM and Mosfet.

ZebbeBa, 2023-03-26:
The Bluepill and Freejoy have a sweet feature that allows users to turn a power circuit on and off using a PWM and a dedicated mosfet or transistor.

This can be used to power backlighting circuits or RGB leds- 
One button is assigned to turn the PWM signal high or low and allowing users to turn the backlighting on and off.

The benefit is that you can use "normal" led instead of WS2812B and are not depended on any clinet software.  
And the circuitry is very simple. 

Freejoy Wiki article on the topic: 
https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/RGB-LED-configuration.md

[image]

andreasdahl1987, 2023-03-26:
Great suggestion, I'll do some experiments and eventually send you something for testing :) As you said, this is very light on hardware/software, and could be extremely useful - so no good reason to not include it.

andreasdahl1987, 2024-04-19:
Added in 2.9

closed 2024-04-19 by andreasdahl1987