andreasdahl1987/DahlDesignDDC #23: Support for rotary encoder EC16

Rekiem24, 2023-03-26:
Hello!

This is not an issue, just a request. (sorry, if this is not the place to add it)

Would be possible to add support for rotary encoder EC16? I tried the different options to see if it was working as it looks similar to EC11, but the best I got is rotating any directions was pushing same button.

https://es.aliexpress.com/item/1005002009833799.html?spm=a2g0o.productlist.main.1.72a338a4MV43Vz&algo_pvid=1b740a8c-73cb-4875-bfa1-6bf319fc3bb6&aem_p4p_detail=202303261306413562080063708700020359545&algo_exp_id=1b740a8c-73cb-4875-bfa1-6bf319fc3bb6-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000018399609239%22%7D&pdp_npi=3%40dis%21EUR%2110.36%218.49%21%21%21%21%21%402100ba4716798612013677128d0712%2112000018399609239%21sea%21ES%21734421856&curPageLogUid=agtK3gK9wca2&ad_pvid=202303261306413562080063708700020359545_1&ad_pvid=202303261306413562080063708700020359545_1

Thanks

andreasdahl1987, 2023-03-26:
Did you try the PEC11 function? Looks like the EC16 is 1 pulse per detent, so that should be the best match.

andreasdahl1987, 2023-03-26:
And yes, this is where you ask for new features 😄

Rekiem24, 2023-03-27:
Hi! Thanks.

I manage to make it work, as I realise that this encoder has the common pin on the right side.

[image]

so PEC11 and rotary2Pulse are working. However with PEC11 sometime miss a pulse and rotary2Pulse works perfect.

Would be possible to create the function for rotary2PulseBite?

I was checking the code....see if I could copy and create this new function...but at the moment still not in that level :(

Thanks
Alex

andreasdahl1987, 2023-03-27:
That is an interesting placement of the common pin indeed! Also interesting that it works so well with rotaryPulse. I only know one other encoder that works that way, and it is not very common. I'll add this encoder to the list of rotaryPulse - encoders in the manual, thanks. I'll add rotaryPulseBite() in the next patch :)

andreasdahl1987, 2023-05-06:
Made rotaryPulseBite() and rotaryPulseBrightness that will be in the comming patch :)

Rekiem24, 2023-05-07:
Awesome!

Thanks! Time to test ;)

andreasdahl1987, 2023-05-07:
Yes you can download the master branch and give it a go :)

andreasdahl1987, 2023-05-07:
Added in #30

closed 2023-05-07 by andreasdahl1987

Rekiem24, 2023-06-25:
Hi,

I've tried the new updated function rotaryPulse() and is not working same way as the previous rotary2Pulse()

Or at least to me rotary2Pulse() is working quite good with encoder EC16, however the new rotaryPulse() is activating both buttons, no matter if you turn left or right.

I hope this helps

Thanks

andreasdahl1987, 2023-06-25:
This is very interesting. The encoder this function was made for is quite unique, and works very different compared to the popular elma, alps, cts and bourne encoders. I rewrote the algorithm to add more functionality for this encoder, and it is working as expected. I was surprised incthe first place that the EC16 works with this function. Going on holiday now, but you can easily copy the old function and replace it in the new DDC if you want to use the latest version. In the meantime Ill have to get my hands on a EC16 amd find out how it works:D