andreasdahl1987/DahlDesignDDC #33: Eror code 10 when finish setup DDC

artayasa, 2023-12-10:
Hi i get error code 10 cannot start, i dont know what happen, but when i try with example sketch keyboard or mouse the HID are normaly. But when i upload DDC frimeware will be error cod 10 again. Maybesome one can telling me

andreasdahl1987, 2023-12-15:
Hey! This is unlikely to have anything to do with DDC. Have you tried googling this error and going through the recommended fixes?

artayasa, 2023-12-15:
> Hey! This is unlikely to have anything to do with DDC. Have you tried googling this error and going through the recommended fixes?

Yess that my mistake, sorry, but for now my encorder not working sir, before working, i try my first code but still not working, and some of my button example button 1, when i press and check on the pc it will be press button 1 and 5 sirr.

andreasdahl1987, 2023-12-15:
Please join the discord: discord.gg/49btqWNmCc

Post your wiring and code in the #help channel and we'll figure it out :)