andreasdahl1987/DahlDesignDDC #42: RP2040 Freezes on Flash EEPROM Write Operation

madlajos, 2025-03-29:
When attempting to use the RP2040’s internal flash as EEPROM, the MCU freezes (hangs) immediately during the flash write (commit) operation. I've tried moving the commit() function (and its helpers) to RAM, but the freeze persists (although I am a beginner with memory mapping). 

I've tried setting clutch calibration settings and bitepoint.

Any insights would be appreciated. Let me know if I can provide any further info/tests.

andreasdahl1987, 2025-04-12:
Im finding it hard to reproduce the issue :/ Which board is this? Can you try with the latest 2.12.1? I added a tiny delay after the commit in case there is anything being interrupted by the flash write.

IvanLaorden, 2025-05-01:
Hi
I'm having the same problem here. Everything works fine, but when I activate the EEPROM for my Raspberry Pi Pico, it freezes when I use the bite button or clutch calibration.
In the case of calibration, it freezes when I press the button after pressing the clutches for calibration. In the case of the bite point, it freezes when I release the button after adjusting the bite point percentage with the encoders.
I've used two different Raspberry Pi Pico cards, one original and one from AliExpress, with the same result. Here's a link: the 16MB card, "RP2040 TYPE-C 16MB" (https://es.aliexpress.com/item/1005006055009344.html?spm=a2g0o.order_detail.order_detail_item.5.2a1339d3twd38P&gatewayAdapt=glo2esp).

I'm not sure if this information will be of any use to you.

If you need more info or to test, I'm here.

I used Google Translate; my English is terrible, sorry.

Thanks.

IvanLaorden, 2025-05-02:
Good evening.
I've been doing some tests, and after trying a lot of things, I realized it was the ADS1115 chip I had installed for the clutches to get better resolution.
I was also having problems with the LEDs. Once I configured all the switches, I put the LEDs in, and it crashed.
I have no idea why, but once I removed that chip, everything works, including the EEPROM.

I'm doing something wrong, but I don't know what. Any idea what I'm doing wrong?

Thanks for all the hard work.

andreasdahl1987, 2025-05-02:
> Good evening. I've been doing some tests, and after trying a lot of things, I realized it was the ADS1115 chip I had installed for the clutches to get better resolution. I was also having problems [...]

Ok, this is interesting. Can you upload your whole sketch folder so I can have a look at your ADS1115 setup and usage?

IvanLaorden, 2025-05-02:
Thanks for your response.
The truth is, I'm having a lot of trouble with this project. From the aforementioned issues to multiple keystrokes, and I have diodes installed. I'm rethinking everything a bit; as soon as I have a new version, I'll upload it. I've changed the entire sketch to continue testing and fix the multiple keystrokes, but so far without success, so I'm going step by step to try to figure out what I'm doing wrong.

Thanks again, and sorry for my broken English.

andreasdahl1987, 2025-05-02:
@IvanLaorden No worries! Ghosting with matrix wiring is really difficult to debug, so I wish you good luck 😅 . Make sure there is no path to/short to ground within your matrix.

IvanLaorden, 2025-05-02:
I already have something working; it's not perfect, but it's working.
I'm having a double-click issue with the switch (6,1), but I've installed it in another one and it works.
I'm uploading the files that work for me with the ADS1115 chip configuration, but without activating it. I'm including a link to the chip I'm using, but I have it working in another project without any issues. I have a different one if you'd like me to test it.
If you need more files, let me know. I'll upload the Excel file with the connection template.
If you prefer, we can continue this topic on Discord, which might be more convenient for you.

Thanks again.

ADS1115 
https://es.aliexpress.com/item/1005006365593004.html?spm=a2g0o.order_list.order_list_main.5.28ab194d8OSVBb&gatewayAdapt=glo2esp
Version not using
https://es.aliexpress.com/item/1005006074781549.html?spm=a2g0o.order_list.order_list_main.11.28ab194d8OSVBb&gatewayAdapt=glo2esp

[HC57 DDC - Hoja 1.pdf](https://github.com/user-attachments/files/20015575/HC57.DDC.-.Hoja.1.pdf)

[02_Board.ino.pdf](https://github.com/user-attachments/files/20015651/02_Board.ino.pdf)
[10_TableAndAnalog.ino.pdf](https://github.com/user-attachments/files/20015653/10_TableAndAnalog.ino.pdf)
[30_Switches.ino.pdf](https://github.com/user-attachments/files/20015652/30_Switches.ino.pdf)

I'm sorry but I don't know how to upload the .ino file directly.

Thanks

andreasdahl1987, 2025-05-02:
@IvanLaorden Let's keep this here until we know for sure it isn't related to the original issue post. Can you confirm:
- Was the ADS1115 ever working with your DDC? Can you get a reading from it?
- Does the FW stop freezing if ADS1115 is active and RP2040EEPROM is disabled?

Can you also share the schematic? At least the part with the ADC and its connecting to the MCU

IvanLaorden, 2025-05-02:
- If it works without problems and you can read with the Arduino IDE serial monitor and use the double clutch but without calibration, with the reading values only.
- Yes, I can even activate it, but when calibrating the bite point, it locks up. Pressing the bite point lock again doesn't lock up without activating the EEPROM.
-These are the connections I have now. When I first connected it, it was connected to a continuous cable for both the GND and the voltages, but since I was having problems, I switched everything to a single cable just in case that was the problem.
I've checked the voltages and don't see anything unusual.
The LED connections are 5.1V.
The clutch connections are 3.27V.
ADS1115 chip is 3.27V.
I've attached a photo of how I have it, so you can see how clumsy I am. I'm sure I have some connections wrong, and they're causing everything to fail.

Sorry for the schematics, I'm a newbie.

[image]

[image]

IvanLaorden, 2025-05-24:
Good afternoon Andreas

I've been testing for days, and I don't understand why, but I can't get rid of phantom inputs.
It's very strange. I plan to use two funky switches and five encoders in my project,( I'll leave a link in case I use a bad configuration) and 21 Omron B3Fs.
Since I can't make any progress with this, I've only tried the other ADS1115 chip and haven't changed anything.

This configuration work, but is soo strange, but some times have ghost inputs, maybe conection problems? I put 99 for ghost inputs

I'm leaving the configurations for both sheets 10 and 30.

Encoders Link

[https://pokornyiengineering.com/products/ec11-300g-encoder?_pos=1&_fid=411ec26e9&_ss=c](url) 

10_TableAndAnalog.ino

#if(USING_CB1 == 0)

//-----------------------------------------
//------------TABLE VARIABLES-------------
//-----------------------------------------

uint8_t row[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8};
const uint8_t rowCount = sizeof(row) / sizeof(row[0]);

uint8_t col[] = { 9, 10, 11, 12, 13, 14, 15};
const uint8_t colCount = sizeof(col) / sizeof(col[0]);

//---------------------------------------
//--------TABLE DESCRIPTION--------------
//---------------------------------------

uint8_t buttonNumber[rowCount][colCount] =
{
  { 0,  1,  2,  3,  4,  5,  6},    //ROW1
  { 7,  8,  9, 10, 11, 12, 13},    //ROW2
  {14,  0,  0, 18, 19,  0,  0},    //ROW3
  {23,  0,  0, 27, 28,  0,  0},    //ROW4
  {32,  0,  0, 36, 37, 38, 99},    //ROW5
  {40, 41, 42, 43, 44, 45, 99},    //ROW6
  {47, 48, 99, 99, 99, 99, 99},    //ROW7
  {99, 99, 99, 99, 99, 59, 60},    //ROW8
  {61, 62, 63, 64, 65, 66, 67}     //ROW9
};

//---------------------------------------
//------------DIRECT INPUTS--------------
//---------------------------------------

uint8_t directPins[] = { 99 };
const uint8_t directPinsCount = sizeof(directPins) / sizeof(directPins[0]);

//---------------------------------------
//------------DIRECT OUTPUTS-------------
//---------------------------------------

#define enableOutput 0
uint8_t outputPins[] = { 99};
const uint8_t outputPinsCount = sizeof(outputPins) / sizeof(outputPins[0]);

//---------------------------------------
//--------ANALOG CHANNEL SETUP-----------
//---------------------------------------

#define analogChannelCount 1                             //Number of analog channels

uint8_t analogPins[analogChannelCount] =                 //Pin number for your analog channels
{ 99 };

uint8_t analogButtonNumber[analogChannelCount] =         //Button number for your analog channels
{ 0 };

uint8_t analogButtonNumberIncMode[analogChannelCount] =  //Button number for alternate modes
{ 0 };

#endif

30_Switches.ino

   //--------------------------------------
  //---------SWITCHES START HERE----------
  //--------------------------------------

  // Push Buttons

  pushButton(2,1);
  pushButton(2,2);
  pushButton(2,3);
  pushButton(2,4);
  pushButton(2,5);
  pushButton(2,6);
  biteButtonLatch(3,4);
  clutchCalibrate(4,4);
  modButton(5,4);
  pushButton(5,5);
  pushButton(5,6);
  pushButton(6,1); 
  pushButton(6,2);
  pushButton(6,3);
  pushButton(6,4);
  pushButton(6,5);
  pushButton(6,6);
  pushButton(7,1);
  pushButton(7,2);
  pushButton(8,7);
  pushButton(8,6);

  // Encoders

  PEC11Stack(3, 3, 1, 2, 3, 1, false);  //ENCODER PULGAR IZQUIERDO
  
  PEC11Stack(3, 7, 2, 2, 3, 5, false);  //ENCODER FRONTAL IZQUIERDO

  PEC11Stack(4, 3, 3, 2, 4, 1, false);  //ENCODER FRONTAL CENTRAL

  PEC11Stack(4, 7, 4, 2, 4, 5, false);  //ENCODER FRONTAL DERECHA

  PEC11Stack(5, 3, 5, 2, 5, 1, false);  //ENCODER PULGAR DERECHO

  // Funkys

  funkyButton(1,1,5,2,3,4);
  funkyButton(1,2,5,1,3,4);
  funkyButton(1,3,5,1,2,4);
  funkyButton(1,4,5,1,2,3);
  funkyPush(1,5,1,2,3,4);
  funkyRotary(1,6,7, false);

  funkyButton(9,1,5,2,3,4);
  funkyButton(9,2,5,1,3,4);
  funkyButton(9,3,5,1,2,4);
  funkyButton(9,4,5,1,2,3);
  funkyPush(9,5,1,2,3,4);
  funkyRotary(9,6,7, false);

  //--------------------------------------
  //---------SWITCHES END HERE------------
  //--------------------------------------

	Joystick.setZAxis(rotaryField - 32767);
	Joystick.setYAxis(buttonField - 32767);

	Joystick.sendState();

} //LOOP ENDS HERE

#if (BOARDTYPE == 2)
  void loop()
  {
	#if (LED1COUNT + LED2COUNT + LED3COUNT + LED4COUNT > 0)
		processCommands();
	#endif
  #if(USING_CB1 == 1 || ENABLE_OVERSAMPLING == 1)
    CB1Oversampling();
  #endif
  }
#endif

Maybe Protoboard Problem? Conections problems

[image]

Any advice on configuring the connection table, another type of protoboard, or another testing system... whatever comes to mind, I've really run out of ideas. I try to change DDC version too, but the same result

Perdón por molestarte con eso.

andreasdahl1987, 2025-05-24:
> Good afternoon Andreas
> 
> I've been testing for days, and I don't understand why, but I can't get rid of phantom inputs.
> It's very strange. I plan to use two funky switches and five encoders in my project,( I'll leave a link in case I use a bad configuration) and 21 Omron B3Fs.
> Since I can't make any progress with this, I've only tried the other ADS1115 chip and haven't changed anything.
> 
> This configuration work, but is soo strange, but some times have ghost inputs, maybe conection problems? I put 99 for ghost inputs
> 
> I'm leaving the configurations for both sheets 10 and 30.
> 
> Encoders Link
> 
> [https://pokornyiengineering.com/products/ec11-300g-encoder?_pos=1&_fid=411ec26e9&_ss=c](url) 
> 
> 10_TableAndAnalog.ino
> 
> #if(USING_CB1 == 0)
> 
> //-----------------------------------------
> //------------TABLE VARIABLES-------------
> //-----------------------------------------
> 
> uint8_t row[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8};
> const uint8_t rowCount = sizeof(row) / sizeof(row[0]);
> 
> uint8_t col[] = { 9, 10, 11, 12, 13, 14, 15};
> const uint8_t colCount = sizeof(col) / sizeof(col[0]);
> 
> //---------------------------------------
> //--------TABLE DESCRIPTION--------------
> //---------------------------------------
> 
> uint8_t buttonNumber[rowCount][colCount] =
> {
>   { 0,  1,  2,  3,  4,  5,  6},    //ROW1
>   { 7,  8,  9, 10, 11, 12, 13},    //ROW2
>   {14,  0,  0, 18, 19,  0,  0},    //ROW3
>   {23,  0,  0, 27, 28,  0,  0},    //ROW4
>   {32,  0,  0, 36, 37, 38, 99},    //ROW5
>   {40, 41, 42, 43, 44, 45, 99},    //ROW6
>   {47, 48, 99, 99, 99, 99, 99},    //ROW7
>   {99, 99, 99, 99, 99, 59, 60},    //ROW8
>   {61, 62, 63, 64, 65, 66, 67}     //ROW9
> };
> 
> //---------------------------------------
> //------------DIRECT INPUTS--------------
> //---------------------------------------
> 
> 
> uint8_t directPins[] = { 99 };
> const uint8_t directPinsCount = sizeof(directPins) / sizeof(directPins[0]);
> 
> 
> //---------------------------------------
> //------------DIRECT OUTPUTS-------------
> //---------------------------------------
> 
> #define enableOutput 0
> uint8_t outputPins[] = { 99};
> const uint8_t outputPinsCount = sizeof(outputPins) / sizeof(outputPins[0]);
> 
> 
> //---------------------------------------
> //--------ANALOG CHANNEL SETUP-----------
> //---------------------------------------
> 
> #define analogChannelCount 1                             //Number of analog channels
> 
> uint8_t analogPins[analogChannelCount] =                 //Pin number for your analog channels
> { 99 };
> 
> uint8_t analogButtonNumber[analogChannelCount] =         //Button number for your analog channels
> { 0 };
> 
> uint8_t analogButtonNumberIncMode[analogChannelCount] =  //Button number for alternate modes
> { 0 };
> 
> 
> 
> #endif
> 
> 
> 30_Switches.ino
> 
>    //--------------------------------------
>   //---------SWITCHES START HERE----------
>   //--------------------------------------
> 
>   // Push Buttons
> 
>   pushButton(2,1);
>   pushButton(2,2);
>   pushButton(2,3);
>   pushButton(2,4);
>   pushButton(2,5);
>   pushButton(2,6);
>   biteButtonLatch(3,4);
>   clutchCalibrate(4,4);
>   modButton(5,4);
>   pushButton(5,5);
>   pushButton(5,6);
>   pushButton(6,1); 
>   pushButton(6,2);
>   pushButton(6,3);
>   pushButton(6,4);
>   pushButton(6,5);
>   pushButton(6,6);
>   pushButton(7,1);
>   pushButton(7,2);
>   pushButton(8,7);
>   pushButton(8,6);
> 
>   // Encoders
> 
>   PEC11Stack(3, 3, 1, 2, 3, 1, false);  //ENCODER PULGAR IZQUIERDO
>   
>   PEC11Stack(3, 7, 2, 2, 3, 5, false);  //ENCODER FRONTAL IZQUIERDO
> 
>   PEC11Stack(4, 3, 3, 2, 4, 1, false);  //ENCODER FRONTAL CENTRAL
> 
>   PEC11Stack(4, 7, 4, 2, 4, 5, false);  //ENCODER FRONTAL DERECHA
> 
>   PEC11Stack(5, 3, 5, 2, 5, 1, false);  //ENCODER PULGAR DERECHO
> 
> 
> 
>   // Funkys
> 
>   funkyButton(1,1,5,2,3,4);
>   funkyButton(1,2,5,1,3,4);
>   funkyButton(1,3,5,1,2,4);
>   funkyButton(1,4,5,1,2,3);
>   funkyPush(1,5,1,2,3,4);
>   funkyRotary(1,6,7, false);
> 
>   funkyButton(9,1,5,2,3,4);
>   funkyButton(9,2,5,1,3,4);
>   funkyButton(9,3,5,1,2,4);
>   funkyButton(9,4,5,1,2,3);
>   funkyPush(9,5,1,2,3,4);
>   funkyRotary(9,6,7, false);
> 
> 
> 
>   //--------------------------------------
>   //---------SWITCHES END HERE------------
>   //--------------------------------------
> 
> 	Joystick.setZAxis(rotaryField - 32767);
> 	Joystick.setYAxis(buttonField - 32767);
> 
> 	Joystick.sendState();
> 
> } //LOOP ENDS HERE
> 
> #if (BOARDTYPE == 2)
>   void loop()
>   {
> 	#if (LED1COUNT + LED2COUNT + LED3COUNT + LED4COUNT > 0)
> 		processCommands();
> 	#endif
>   #if(USING_CB1 == 1 || ENABLE_OVERSAMPLING == 1)
>     CB1Oversampling();
>   #endif
>   }
> #endif
> 
> 
> Maybe Protoboard Problem? Conections problems
> 
> 
> 
> [image]
> 
> Any advice on configuring the connection table, another type of protoboard, or another testing system... whatever comes to mind, I've really run out of ideas. I try to change DDC version too, but the same result
> 
> Sorry for disturb you with that
> 

Hey! I don't think this is related to the original post, better type this in the #help channel in my Discord. But this is likely due to missing diodes. You're not using diodes, right? You can't really have encoders in the matrix if you're not using diodes

IvanLaorden, 2025-05-26:
Hi Andreas,

Thanks for the advice, newbies need more space, hahahahaha
That's what I've been doing. I've been testing for days when it seemed like it would work, but when I put the funky ones on, the problems started, or even with the rotary ones.
But now I have it. It's true that I'm still having some problems. If you want, I'll tell you about it here or post it on Discord, whichever you prefer.
In principle, everything is working, both the clutches and the LEDs, albeit in a basic way. I just activated the EEPROM and was able to calibrate the clutches from the calibration button. On a couple of occasions, it didn't pick them up correctly. I think it's because I was pressing both at the same time. But after activating them separately, it picked them up correctly.
Now I have to go to work. I'll continue testing tonight, but in principle, I've already managed to do more than the last time.
I think the problem was mine. I'm not sure exactly what it was. It could be an incorrect board configuration in the Arduino IDE, but after dozens of tests, I can't say for sure. What I did change was the firmware version. Instead of using the one listed for download, 2.12.1, released on April 12th, I used the master version. I don't know if I understand.

I'll keep you posted.

Thanks for everything.
Especially for your great work and dedication.
Let's talk.

Google translate. Sorry.

IvanLaorden, 2025-05-27:
Hi  Andrea,

I finally have it. Everything I've tested works. I confirm I'm using version 2.13, which doesn't reconnect the SimHub DDC plugin, but that doesn't matter to me now, as it's a lot of work to get my hands on the wheel.
I'll try version 2.12.1 again this weekend, but I suspect it will work. I'll let you know when I check it out.

Thanks for all

andreasdahl1987, 2025-05-28:
> Hi  Andrea,
>
> I finally have it. Everything I've tested works. I confirm I'm using version 2.13, which doesn't reconnect the SimHub DDC plugin, but that doesn't matter to me now, as it's a lot of [...]

Glad to hear! What do you think was the issue?

IvanLaorden, 2025-05-28:
I'd bet on a poor choice of Arduino IDE on the wrong board, but I'm not sure. Or maybe it was because all the plugging and unplugging of the DuPont connectors wasn't making good contact, although I don't think so. After so many tests, I don't even remember what I was doing, hahaha.
I want to try different things with this configuration, since this one works, and see if compiling with another board and/or version 2.12.1 fails. So next time, I hope I don't screw up again.
While I'm printing the steering wheel parts to test that everything fits, I'll keep myself busy with this. That way, I don't get the urge to start ordering parts and then find they don't work, hahaha.

I'll let you know as soon as I know something.

Thanks, Andreas.

andreasdahl1987, 2025-06-14:
> Good evening
> 
> Sorry for the delay, but I had to fix some things on my prototype steering wheel that took longer than expected.
> I tested version 2.12.1 with exactly the same configuration as 2.13, and it works perfectly.
> I changed the row of one of the funky switch encoders because it was causing me to double-click one of the B3Fs, but only on alternate clicks of the funky switch encoder. I don't know if I understand. But I tried version 2.13, and the same thing happened.
> It no longer crashes, even with the EEPROM enabled; everything works fine.
> So, it was most likely my problem.
> Sorry if I bothered you.
> In principle, I'm moving forward with my project. Now it's time for the PCB. 
> First, learn how to make them, hahaha.
> Any tips or tricks on how to order the layers so I don't have problems?
> 
> Thanks for everything, and sorry again.

Happy to hear you fixed it:D have a look at this guys channel, a lot of people are learning from him: https://youtu.be/aODkA2mrimQ?si=Q21BrTu8QS_4UQcs

IvanLaorden, 2025-06-20:
Thanks Andreas,

Working on it

[image]
[image]

I'm scared of this project, and I'm not even going to tell you about the PCB, hahaha

Thanks for everything Andreas

andreasdahl1987, 2025-08-04:
Still no clue what is causing this for you @madlajos , but Ill keep the report open