andreasdahl1987/DahlDesignDDC #43: Switch table special functions

Edermg05, 2025-05-01:
First of all, I would like to thank you for the development of this firmware and for keeping it open-source. (By the way, I don't know if this issue is suitable for this repo. Feel free to remove it or tell me where to post it)

So, I am obviously trying to develop a DIY wheel, and the best thing I knew was "FreeJoy". A couple of days ago I discovered DDC, and I am having a blast with the amount of things it can do and the depth of customization it offers.

I have read through all the GitBook "wiki" and have also taken a look at the different repositories and the arduino sketches (and some of your yt videos aswell), but I haven't found any place in which there is an explanation for the "Special function" assignment for the inputs.

I obviously understand some of them, like neutral, bite point, etc. But, I'm genuinely struggling with "DD Button", "Hybrid button", "Modifier button" (I understand it allows you to access a second layer of button assignments while keeping it pressed or similar?) or the "Quick/Magic switch".

Am I missing something obvious or any ReadME where there is an explanation for these functions? In that case, I apologise, but haven't managed to find it.

Anyways, I just would like to find out a little bit more regarding this. Thank you in advance, any answer is appreciated.

andreasdahl1987, 2025-05-26:
I'll add a page under 1.Project planning explaining these features. As pointed out, the way the manual is currently written you'll have to know how these things work in order to find the examples on how to use them

Edermg05, 2025-05-28:
> Hi, I believe [THIS](https://dahl-design.gitbook.io/ddc/switch-library/function-button) page on the GitBook wiki might shed some more light on what each button function does. Unfortunately, they aren't helpful unless you are already using the features that they modify

Hi, thank you for your answer. I ended up asking the same on discord and found out about this page, so thank you for taking the time to answer

andreasdahl1987, 2025-08-04:
Documentation added in 2.13

closed 2025-08-04 by andreasdahl1987